disk91/stm32-it-sdk #26: MAX17201 dev name register reading wrong

martincornu, 2020-03-30:
Hi Disk91,

The MAX17201 has a wrong dev name default value in register and it causes a MAX17205_NOTFOUND at setup of the driver.

It seems that is because the configuration must be done first of all in the PackCfg (0x0bd) register. For example: write 0x3a03 for a 3 cells battery pack (0x3a0N for a multi-cell battery pack, N is the number of cells).

I found that thanks to this [post](https://electronics.stackexchange.com/questions/426264/max17205-devname-reading-wrong-but-other-default-value-is-correct-why).

disk91, 2020-03-30:
Thank you martin for reporting this !

disk91, 2020-03-30:
I propose to fix it like this.	
Does it match with your test martin ?

__max17205_config.initialized = MAX17205_FAILED;

	// Preconfiguration related to MAX17205 bug
	// see https://github.com/disk91/stm32-it-sdk/issues/26
	switch ( mode ) {
		case MAX17205_MODE_3CELLS_INT_TEMP:
			// 3 cells 
			__writeRegister(ITSDK_DRIVERS_MAX17205_REG_BNPACKCFG_ADR,0x3A03);
			break;
		default:
		case MAX17205_MODE_DEFAULT:
			break;

		}

	
	// Search for the device type
	uint16_t v;
	if ( __readRegister(ITSDK_DRIVERS_MAX17205_REG_DEVNAME_ADR,&v) != I2C_OK ) {
		ITSDK_ERROR_REPORT(ITSDK_ERROR_DRV_MAX17205_NOTFOUND,0);

disk91, 2020-03-30:
Can you confirm it work with this seting ?

martincornu, 2020-03-30:
Yes that fix works well I have tested it ;) Thanks!

disk91, 2020-03-30:
Thank you for the feedback, the merge has been made in devel branch

closed 2020-03-30 by disk91