disk91/stm32-it-sdk #27: ST25DV low power in setup

martincornu, 2020-04-07:
Hi disk91,

My problem is that when I call the st25dv setup function and then want my device to sleep the st25dv does not go to low power mode and has a high power consumption (+20uA).

In the st25dv.c driver it seems that the function drivers_st25dv_goLowPower() at the end of the setup function could never enter the if statement because the __st25dv_config.state always equals ST25DV_SLEEPING.

Thanks,
Martin

disk91, 2020-04-07:
You are fully right Martin. Thank you for the feedback.
This has been fixed in commit https://github.com/disk91/stm32-it-sdk/commit/f73be60d02a3f2e4bebdc42bb8592bd48326c681 in the devel branch.
Please confirm it works

martincornu, 2020-04-07:
Yes that's good! Thanks.

closed 2020-04-07 by martincornu