disk91/stm32-it-sdk #40: Build project atolic

GreenChicken123, 2020-09-13:
Hi Paul, 
Thank you very much for providing an alternative to the ST example code, great work ! 
However I've been encountering a few issues in building the project. I am working using TrueStudio to be as close as possible to the eclipse example you showed in the YT video. 
I followed step by step the project set-up but the project build fails with the following errors:

'ITSDK_DEFAULT_NETWORK' undeclared (first use in this function)	sdk_config.c	/board-lora/stm32-it-sdk/Src/it_sdk/eeprom	line 72	C/C++ Problem
'ITSDK_DEFAULT_REGION' undeclared (first use in this function)	sdk_config.c	/board-lora/stm32-it-sdk/Src/it_sdk/eeprom	line 81	C/C++ Problem

Any idea where I went wrong ? 
Thanks in advance for your help  !

Maxime

disk91, 2020-09-13:
Hello @GreenChicken123, 
Apparently you are using one of the example repository, made with a specific (and previous) build. The REAME.md of the example project will tells you what build is compatible.
On the other hand it's always better to use the last build. For this you need to adapt the config.h file to add the required new definitions.
```C
#define ITSDK_DEFAULT_NETWORK		__ACTIV_NETWORK_NONE					// Default network to activate

//#define ITSDK_DEFAULT_NETWORK		__ACTIV_NETWORK_SIGFOX					// Default network to activate
//#define ITSDK_DEFAULT_NETWORK		__ACTIV_NETWORK_LORAWAN					// Default network to activate
#define ITSDK_DEFAULT_REGION		__LPWAN_REGION_EU868				    // default region to activate 
```
but some other could be missing so it's better to review the configurations file with the template version. The it-sdk readme file also try to list the added defines version after versions

GreenChicken123, 2020-09-13:
Thank you for your very fast answer ! 
 If I understand correctly, the example project for the STM32 is no longer up to date. Would it be better (cleaner ?) to start a fresh project with cube MX and include the SDK on the side ? 

Regarding my previous post, it seems that in TrueStudio, removing the source path in the project properties is not enough to unlink the file from the build. They also need to be removed from the project explorer. Project is compiling now thanks !

disk91, 2020-09-13:
You welcome, thank you for the feedback

closed 2020-09-13 by disk91