disk91/stm32-it-sdk #47: Not able to use GPIO event

zokl, 2020-09-23:
Hello, I try to use an interrupt with gpio button. I used followed (snippet) code but after pushing the button MCU after 5 seconds restarts. In my code, I increment the global variable attemp. 
Do you know where is a problem?

Thank you very much for your answer.

```
#define BUTTON_PORT	__BANK_B
#define BUTTON_PIN __LP_GPIO_2
uint16_t attemp;

static void __button_interrupt(uint16_t GPIO_Pin) {
	attemp++;
}

static gpio_irq_chain_t __button_gpio_irq1 = {
		__button_interrupt,
		__LP_GPIO_2,
		NULL,
};

void project_setup() {
	gpio_interruptDisable(BUTTON_PORT, BUTTON_PIN);
	gpio_interruptClear(BUTTON_PORT, BUTTON_PIN);
	gpio_configure(BUTTON_PORT, BUTTON_PIN, GPIO_INTERRUPT_RISING_PULLDWN);
	gpio_interruptPriority(BUTTON_PORT, BUTTON_PIN, 2, 0);
	gpio_registerIrqAction(&__button_gpio_irq1);
	gpio_interruptEnable(BUTTON_PORT, BUTTON_PIN);
	attemp = 0;
}
```

zokl, 2020-09-23:
I am so sorry. It was my fault. I forgot add HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_2); to IRQ Handler. 

Thank you for your good work!

closed 2020-09-23 by zokl

disk91, 2020-09-23:
Sure you need this, I may improve the doc on this.
The other problem you could have in a such situation is not protecting the pin against being switched analog on low power mode. This is made in config.h but KEEPing the pin unchanged during sleep.

zokl, 2020-09-23:
Yes, you have true. Updating the doc will be good. For me is not good to understand how gpio in low power works. I am trying to use GPIO interrupt from low power mode but without success. I would like to have a button and after pushing it make a code in a callback.