disk91/stm32-it-sdk #55: Dynamic sleep time

zokl, 2020-10-31:
Hello,

I would like to ask you, what is the correct procedure if I want to change the sleep time over time?

Now I am trying to use a scheduler: _itdt_sched_registerSched (300000, ITSDK_SCHED_CONF_IMMEDIATE and getPower)_ and I have to change my sleep time over time. I didn't find any way to unregister and register again the scheduler.

Thank you,
Zbynek Kocur

disk91, 2020-10-31:
Currently the way the task manager is develop it does not support task removal. 
The recommended implementation way is more to have a periodic wakeup and a state machine to manage different behaviors related to the device step. regarding the long sleep period you want the power consumption cost sounds low.

Btw, You proposal is interesting and I keep it in the backlog. I need to find a way to make it without making the code bigger. I think a way could be to have the ability to change the period of a halted Sched. So I propose to add un function itdt_sched_changePeriod(uint8_t schedId, uint32_t periodMs) you can execute on a previously halted sched.

Would that fit with your request ?

zokl, 2020-11-01:
> Currently the way the task manager is develop it does not support task removal.
> The recommended implementation way is more to have a periodic wakeup and a state machine to manage different behaviors [...]

Yes, this kind of function sound good and will be usefull for longer sleep periods. 

In the other hand. I try to maintain ITSDK_LOWPOWER_RTC_MS sleep interval by global variable and it works too if I put it at the end of it_sdk_loop. All of my code is in the loop without scheduler.

disk91, 2020-11-05:
Function added in merge #59 to modify the sched parameters. the sched need to be halted as a security before calling for change. it means the next execution will be after the periodMs duration. Callback function can also be changed.

Advantage of scheduler over Sleep period with RTC is you can have different task with different periods and the wakeup will be made according to this variable duration.

closed 2020-11-05 by disk91

zokl, 2020-11-09:
Hello,

I have tried your function `itdt_sched_changeSched` but I have a problem with sleep time. After using the function in the followed way:

    uint32_t sleep = status.sleep_time_s * S_TO_MS;
    itdt_sched_haltSched(sched_id);
    itdt_sched_changeSched(sched_id, sleep, NULL);
    itdt_sched_runSched(sched_id);

The system sleeps twice the time that I set if I disable the condition haltSched and modify only period, sleeping works that I imagine.

Do you know where the problem is?

Thank you, Zbynek Kocur