disk91/stm32-it-sdk #56: Low power consuption

zokl, 2020-11-05:
Hello

I would like to ask you if you have no idea where to look for a problem with the platform sleeping, if during the first sleep it shows consumption of about 3 uA and after wake up and subsequent sleepings, the minimum consumption is 2 mA. He will never get to 3uA from the first sleep. I used MURATA CMWX1ZZABZ with a Sigfox extension and low power mode __LOWPWR_MODE_STOP.

disk91, 2020-11-05:
Hello, basically this is not a SDK issue but an implementation issue. 
- did you create the void stm32l_lowPowerRestoreGpioConfig() function to correctly restore the GPIO pins after waking up ?
- did you verify your wakeup condition to ensure you are not immedialty waking up after sleeping ?

You can enable ITSDK_LOGGER_MODULE -> __LOG_MOD_LOWPOWER to trace the wake reason and find your configuration issue.

Sincerely.

closed 2020-11-05 by disk91

zokl, 2020-11-05:
> Hello, basically this is not a SDK issue but an implementation issue.
>
> * did you create the void stm32l_lowPowerRestoreGpioConfig() function to correctly restore the GPIO pins after waking up ?
> * [...]

Hello,

the first sleep is OK (power consuption about 3 uA) and it sleep until regular wakeup but the second and other sleeps have power consuption about 2 mA. Ther will be a problem with sleeping some parts of mcu. The wake up condition by rtc or gpio is correct. I will try to debug the processes of disabling gpio and cpu at my platform. For waking up i used your scheduler.