disk91/stm32-it-sdk #62: MAX17205

swapmicro, 2020-11-10:
hello there,
I am working with Fuel Guage IC MAX17205, 
My hardware schematic design configuration is as per the primary cell implementation.
The battery voltage is measured from CellX Pin.
The Rsense resistor value is 0.01Ohm (10mOhm)
the reference hardware schematic,
https://www.maximintegrated.com/en/design/technical-documents/app-notes/6/6416.html
[image]
When I use 4 cells with 10mOhm resister on the CellX pin, the default value of the register 0x21h is 0x1105.
When I use 3 cells with 10mOhm resister on the CellX pin, the default value of the register 0x21h is 0x8809.
When I use 4 cells without 10mOhm resister on the CellX pin, the default value of the register 0x21h is 0x0809.
Can you suggest to me, how to configure MAX17205 for 4 cells pack to read the current, voltage, capacity of a battery?
I am using Panasonic AA (1.5V) 4S cells
Thanks
Swapnil

disk91, 2020-11-10:
Hello,
If you want to know how to use the component, I think you should ask to
Maxim itself. If your question is how to modify the code to support this
mode with this component within the SDK please open an issue with the
configuration you would like to have and I'll do my best to integrate it
into the driver.
Sincerely,
Paul

swapmicro, 2020-11-11:
Hello,
Can you please integrate a configuration for a non-rechargeable 4S Cell pack with hibernate mode?
Thanks 
Swapnil

disk91, 2020-11-11:
Sure - if you give me the details of the configuration to be apply I can integrate it into the global code.

Say differently : my role is not to read and understand the long documentation for your own interest. I do such inclusion in as a service.
 
Sorry to be direct, I do my best to focus on feature targeting most of people instead of a single one.