disk91/stm32-it-sdk #68: Not able to use change baudrate on uart2

zokl, 2020-11-24:
I am not able to change a baud rate on usart2 if I use pull mode. Changing is now limited only to IT mode.

```
/**
 * Change the Uart setting baudrate
 * Return BOOL_TRUE on success
 */
itsdk_bool_e serial2_changeBaudRate(serial_baudrate_e bd) {
	UART_HandleTypeDef * lhuart;
	#if  ( ITSDK_WITH_UART_RXIRQ & __UART_USART2 )
	   lhuart = &huart2;
	#else
		return BOOL_FALSE;
	#endif
	switch( bd ) {
		case SERIAL_SPEED_300 : lhuart->Init.BaudRate = 300; break;
		case SERIAL_SPEED_600 : lhuart->Init.BaudRate = 600; break;
		case SERIAL_SPEED_1200 : lhuart->Init.BaudRate = 1200; break;
		case SERIAL_SPEED_2400 : lhuart->Init.BaudRate = 2400; break;
		case SERIAL_SPEED_4800 : lhuart->Init.BaudRate = 4800; break;
		default:
		case SERIAL_SPEED_9600 : lhuart->Init.BaudRate = 9600; break;
		case SERIAL_SPEED_19200 : lhuart->Init.BaudRate = 19200; break;
		case SERIAL_SPEED_38400 : lhuart->Init.BaudRate = 38400; break;
		case SERIAL_SPEED_57600 : lhuart->Init.BaudRate = 57600; break;
		case SERIAL_SPEED_115200 : lhuart->Init.BaudRate = 115200; break;
	}
	serial2_flush();
	if (HAL_UART_Init(lhuart) != HAL_OK) {
	  return BOOL_FALSE;
	}
	serial2_init();
	return BOOL_TRUE;
}
```

disk91, 2020-11-26:
Thank you for reporting and fix it. Your change as been updated.

closed 2020-11-26 by disk91

zokl, 2020-11-27:
Hello, you push the wrong fix of my issue and commit please look here: https://github.com/disk91/stm32-it-sdk/pull/69/files

commit b1934f51 (2020-11-26): Fix #68 - Not able to use change baudrate on usart2