disk91/stm32-it-sdk #73: running on STM32L071

JuergenFitz, 2020-12-13:
I am trying to get your software package STM32 IoT
SDK running on a board I developed myself. (STM32L071CZ + SX1276). The
software is running so far, but I cannot join into the TTN network.
The SX1276 sends the data successfully. I see it on the spectrum
analyzer but the registration does not arrive at the TTN. Another
purchased development board with ESP and Arduino runs successfully
with the same TTN settings. 

[log.txt](https://github.com/disk91/stm32-it-sdk/files/5684831/log.txt)

disk91, 2020-12-13:
Hello, Thank for sharing your experience.
My assumption is your solution does not receive downlink (JoinAccept). It could be related to a misconfiguration in the GPIO/IRQ.
Did you verify you hardware configuration in configLoRaWan.h ?
Did you correctly rewrite void stm32l_lowPowerRestoreGpioConfig() in gpio.c ?
Did you made a try with lowPower mode disabled ? ( call lowPower_disable(); at end of project_setup() )
Let me know about these points for longer investigation.
Paul

disk91, 2020-12-13:
Other question : is your TTN gateway near to the device ? too high signal will jam the response. You can use attenuator on the gateway of pur your hand around the device antenna. It helps in that case.
It would be great to get the timestamp in your log to help verify the timing.
Did you verify the clock frequency ? in config.h
```
#define ITSDK_CORE_CLKFREQ			32000000	
```
must match with your CLK otherwise the timing are not good. If you have something to measure the current consumption like OTii ARC you can verify the timings You should see the reception switch 5s after end of JoinRequest transmission.
Paul

JuergenFitz, 2020-12-13:
I checked the configuration. the frequency is identical in the cubeMx and in config.h 32000000. LowPower_disable (); at the end project_setup () did not bring any improvement. the stm32l_lowPowerRestoreGpioConfig () in gpio.c is adapted. I created the logs with a timestamp. and the current consumption over 10 ohm was recorded with an oscilloscope. I changed the #define ITSDK_LORAWAN_DEFAULT_DR to __LORAWAN_DR_4 as a test. Strangely, with the setting DR4, the program does not stay with ", lorawan_driver_LORA_getJoinState, lorawan_driver_LORA_Join (mode: 0), LoRaMacMlmeRequest return error (11)" as long as with DR0. but maybe it's normal.

the gateway is indeed very close (4M). I covered the antenna and even put it in a metal box. without change.

[serial_console_session_2020-12-13_DR4.txt](https://github.com/disk91/stm32-it-sdk/files/5685268/serial_console_session_2020-12-13_DR4.txt)
[serial_console_session_2020-12-13.log](https://github.com/disk91/stm32-it-sdk/files/5685270/serial_console_session_2020-12-13.log)
[image]
[image]
[source.zip](https://github.com/disk91/stm32-it-sdk/files/5685283/source.zip)

disk91, 2020-12-13:
Apparently the RX timing sounds good.
If you GW is 4M, it's really short and could be a reason, try to move it...

I see you did test with ESP previously, please make sure the DEVEUI / APP EUI are in the right order as on Arduino they are reversed, not in it-sdk. You can see the byte received in the TTN gateway live data and verify your dev eui.

Make sure your deveui and appeui are correctly set by printing them from the console.

JuergenFitz, 2020-12-19:
unfortunately it still doesn't work. app eui and dev eui are correct.
At least the gateway receives a join request and answers with join accept. Unfortunately, my node does not receive the accept. it always ends up in Dio1 interrupt due to RX timeout. It is also interesting that the time between SX1276SetRx and Interrupt Dio1 is very short (approx. 200ms) see picture from the oscilloscope. the transmission time with DR0 is actually significantly longer. Channel 1 DIO0, channel 2 DIO1, channel 3 GPIO for debugging.
First rising edge of channel 3 SX1276SetTx, falling edge of channel 3 enter SX1276SetRx, second rising edge of channel 3 leave SX1276SetRx.
[image]