disk91/stm32-it-sdk #83: Sigfox message sending freeze on MURATA & SX1276

zokl, 2021-06-10:
Hello,

I've run into a problem sending sigfox messages on some of my boards. I am sending a data message with sensor measurements at regular intervals. The sensor wakes up, measures for 3 seconds, and sends the message. 

But several times a day I send messages that were just the battery value. The sensor wakes up, compiles the report in a few hundred milliseconds, and sends. On most boards, it is not a problem, but on some, I have encountered that the sending gets stuck. If I turn on the debug mod on sigfox everything goes as it should, likewise if I put a 500 ms wait before sending this message. It looks to me like the sigfox infrastructure has not been fully initialized. It only does this with some HW pieces though. Any idea what this could be?

It doesn't seem to be a complete system solution to give a 500 ms wait before sending a message, but it works.

disk91, 2021-06-10:
Hello, 
This could be related to a similar issue I've fixed for the MCCI stack related to SPI state not always as expected after a previous communication.
Do you have this commit in your code ? (this is in devel currently)
https://github.com/disk91/stm32-it-sdk/commit/935dcc9b779135b75fc0c4756bb975837fbb4926

This is activated with  ITSDK_MCCI_SPI_HACK define.

Let me know if it fix your problem.

disk91, 2021-06-10:
Second point : the 500 ms is not related to library background initialization as there is no background task. But is could be related to TCXO wakeup time of things like this. Eventually decoupling refresh. 500 ms is really long, did you tried with shorter values ? 
You can try to change the TCXO wakeup time in your sigfox config file to make a test and remove your 500ms delay. 
Let me kow, this problème is really important to capture and fix.
Paul

zokl, 2021-06-10:
> Second point : the 500 ms is not related to library background initialization as there is no background task. But is could be related to TCXO wakeup time of things like this. Eventually decoupling [...]

Hello, time below 400 ms makes a problem. I set 500 ms and it is OK. I will try to increase TCXO wake-up time.

zokl, 2021-06-10:
> Hello,
> This could be related to a similar issue I've fixed for the MCCI stack related to SPI state not always as expected after a previous communication.
> Do you have this commit in your code ? (this is in devel currently)
> [935dcc9](https://github.com/disk91/stm32-it-sdk/commit/935dcc9b779135b75fc0c4756bb975837fbb4926)
> 
> This is activated with ITSDK_MCCI_SPI_HACK define.
> 
> Let me know if it fix your problem.

I will try it too.

zokl, 2021-06-10:
Which variable can I modify? ITSDK_MURATA_WAKEUP_TIME or ITSDK_MURATA_TCXO_WARMUP?

```
#ifndef ITSDK_MURATA_TCXO_WARMUP
#define ITSDK_MURATA_TCXO_WARMUP	50									   // Warmup time for TCXO
#define ITSDK_MURATA_WAKEUP_TIME    53 									   // WakeUp time correction for RX window start at least TCXO Time [ms]
#endif
#define ITSDK_SX1276_SFXWAKEUP_TIME 60 								   	   // WakeUp time correction interframe timing (includes TCXO)
```

disk91, 2021-06-10:
ITSDK_MURATA_TCXO_WARMUP

zokl, 2021-06-10:
> ITSDK_MURATA_TCXO_WARMUP

Looks like a good way to go. On the test board the 100 ms time helped and the upload is working so far. We'll see what other pieces.

zokl, 2021-06-16:
Hello, the TCXO_WARMUP fix works on several boards. Few boards doesn't works. The sending problem occures if the interval between messages is longer then approx 1 hour. If the interval is smaller thank 1 hour everything works good but after one hour the sendings freezes.

Now, I am tryiing your patch from here https://github.com/disk91/stm32-it-sdk/commit/935dcc9b779135b75fc0c4756bb975837fbb4926

zokl, 2021-06-17:
> Hello, the TCXO_WARMUP fix works on several boards. Few boards doesn't works. The sending problem occures if the interval between messages is longer then approx 1 hour. If the interval is smaller thank 1 hour everything works good but after one hour the sendings freezes.
> 
> Now, I am tryiing your patch from here [935dcc9](https://github.com/disk91/stm32-it-sdk/commit/935dcc9b779135b75fc0c4756bb975837fbb4926)

The patch doesn't work for me. I think it could be the HW problem of the selected piece of HW. For many board works longer TCXO WARMUP interval.

Thank you for your help.

zokl, 2021-06-24:
Hello Paul, I would like to reopen this issue because TXCO WARMUP increasing works only for 2h sleeping. If the board sleeps longer, it does not work. For now, I was able to catch the debug output of the freezing part of the communication. This happens after 12 hours of deep sleep (stop mode). Do you have an idea where is the problem?

```
Jun 23 15:25:47 itsdk_sigfox_sendFrame                                                                                                                                                                                                                                                                                            [4/1908]
Jun 23 15:25:47 itsdk_sigfox_setTxPower
Jun 23 15:25:47 itsdk_sigfox_setTxPower_ext(14)
Jun 23 15:25:47 itsdk_sigfox_setTxSpeed
Jun 23 15:25:47 >> SE_NVM_get
Jun 23 15:25:47 >> CREDENTIALS_get_payload_encryption_flag
Jun 23 15:25:47 >> CREDENTIALS_get_dev_id
Jun 23 15:25:47 itsdk_sigfox_getDeviceId
Jun 23 15:25:47 >> CREDENTIALS_aes_128_cbc_encrypt
Jun 23 15:25:47 >> SE_NVM_get_key_type()
Jun 23 15:25:47 >> CREDENTIALS_get_key
Jun 23 15:25:47 itsdk_sigfox_getKEY
Jun 23 15:25:47 >> SE_NVM_set
Jun 23 15:25:47 >> MCU_API_set_nv_mem
Jun 23 15:25:47 >> RF_API_init
Jun 23 15:25:47 RF_API_init in TX
Jun 23 15:25:47 >> STLL_Radio_Init
Jun 23 15:25:47 >> SX1276Init
Jun 23 15:25:47 >> mSX1276SetXO (ON)
Jun 23 15:25:47 >> SX1276Reset
Jun 23 15:25:47 >> RxChainCalibration
Jun 23 15:25:47 >> SX1276SetChannel 868000000
Jun 23 15:25:47 >> SX1276SetChannel 434000000
Jun 23 15:25:47 >> mSX1276SetAntSwLowPower (Off)
Jun 23 15:25:47 >> mSX1276SetXO (OFF)
Jun 23 15:25:47 >> mSX1276IoIrqInit
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetModem (LORA)
Jun 23 15:25:47 >> SX1276SetSleep
Jun 23 15:25:47 >> mSX1276SetAntSwLowPower (Off)
Jun 23 15:25:47 >> mSX1276SetXO (OFF)
Jun 23 15:25:47 >> SX1276SetModem (FSK)
Jun 23 15:25:47 >> SX1276SetSleep
Jun 23 15:25:47 >> mSX1276SetAntSwLowPower (Off)
Jun 23 15:25:47 >> mSX1276SetXO (OFF)
Jun 23 15:25:47 >> SX1276OnDio3Irq
Jun 23 15:25:47 >> mSX1276GetWakeTime
Jun 23 15:25:47 >> STLL_RW 01,80
Jun 23 15:25:47 >> STLL_RW 1D,00
Jun 23 15:25:47 >> STLL_RW 09,7F
Jun 23 15:25:47 >> STLL_RW 3D,A1
Jun 23 15:25:47 >> STLL_RW 36,01
Jun 23 15:25:47 >> STLL_RW 44,7A
Jun 23 15:25:47 >> STLL_RW 4B,19
Jun 23 15:25:47 >> STLL_RW 46,03
Jun 23 15:25:47 >> STLL_RW 0A,0F
Jun 23 15:25:47 >> STLL_RW 70,D0
Jun 23 15:25:47 >> STLL_RW 40,01
Jun 23 15:25:47 >> STLL_RW 41,F1
Jun 23 15:25:47 >> STLL_RadioPowerSetBoard(14)
Jun 23 15:25:47 >> mSX1276SetRfTxPower (14)
Jun 23 15:25:47 RF_API_change_frequency(868129600)
Jun 23 15:25:47 >> SX1276SetChannel 868129600
Jun 23 15:25:47 RF_API_send (1)(26)
Jun 23 15:25:47 >> mSX1276SetXO (ON)
Jun 23 15:25:48 >> mSX1276SetAntSwLowPower (Ready)
Jun 23 15:25:48 >> mSX1276SetAntSw (Tx)
Jun 23 15:25:48    rfo route
Jun 23 15:25:48 >> STLL_SetClockSource:  HSE Selected

```

disk91, 2021-06-27:
Nothing clear about it. 
Did you take a look to serial port ( I did have similar issues due to noise on serial RX ligne during clock switch ) ?
As this kind of problem is really hard to fix, I think the best is to wake up on regular basis a few miliseconds if that help to bypass the issue.